KingAbad/Autonomous_Cart_2 #9: Takenverdelen

kew10, 2020-03-18:


kew10, 2020-03-22:
Ik heb een teammeeting georganiseerd met onze product owner (Frederik) om de taken te verdelen. We hebben onderschat hoeveel wij in een sprint zouden kunnen afmaken, waardoor niet iedereen genoeg had om te doen voor dit sprint. Voor volgende sprint moeten we meer realistische goals opstellen. Ali gaat voorwerken en proberen leds aan te sturen adv sensoren. Ik ga de backlog updaten/ project beschrijving opstellen / metehan help met documentatie motor controller.

ALi-Natiq, 2020-03-22:
ik ga de RGB leds adv ultrasonic sensor aanpakken. dus eerst ga ik proberen om de ultrasonic sensor te programmeren gebruik maken van raspberry pi en daarna ga ik zijn signalen gebruiken om de RGB leds aan te sturen.

meto51, 2020-03-22:
Onderzoek naar motorcontroller en de tekst uitgetypt in een document

meto51, 2020-03-22:
Helpen met de communicatie van de jetson nano

kew10, 2020-03-24:
iedereen moet helpen om issues bij te voegen!

ALi-Natiq, 2020-03-24:
ultrasonic is succesvol geprogrammeerd met de pi

ALi-Natiq, 2020-03-24:
Er moet verder de RGB leds kunnen sturen gebruik maken van ultrasonic sensor.

KingAbad, 2020-03-24:
Voor iedereen was het duidelijk na de meeting van 18/03/2020 wat hij moest doen.

closed 2020-03-27 by Fre101