KingAbad/Autonomous_Cart_2 #15: Kan communiceren met de PLC

kew10, 2020-03-22:


KingAbad, 2020-03-31:
Begonnen met communicatie te regelen met PLC server samen met Oguzhan en Berkan van groep 2

KingAbad, 2020-03-31:
Ik heb eerst ervoor gezorgd dat ik over SSH met de Jetson Nano kan communiceren

KingAbad, 2020-03-31:
IP adres Jetson Nano gevonden met AngryIpScanner

[image]

KingAbad, 2020-03-31:
Nadat ik de IP had gevonden heb ik communicatie gestart met PuTTY

[image]

KingAbad, 2020-03-31:
Succesvolle verbinding met de Jetson Nano over SSH

[image]

KingAbad, 2020-03-31:
Ik heb om te testen een Server en Client python script geschreven om berichten naar elkaar te kunnen sturen en het werkt

[image]

KingAbad, 2020-03-31:
Omdat de PLC Server nog niet volledig klaar is help ik nu degene die daarvoor verantwoordelijk zijn

[image]

KingAbad, 2020-03-31:
PLC Server Volledig klaar maar communicatie werkt niet

KingAbad, 2020-03-31:
Eerstefout gevonden: Niet juiste Poort en Ip adres ingegeven

KingAbad, 2020-03-31:
Tweedefout gevonden: Firewalls stonden aan voor outbound connecties. Dus alle firewalls inclusies windows defender uitgeschakeld

[image]

Een nieuwe outbound rule "NetworkProgramming" aangemaaktt om alle outbound connecties door te laten

[image]

KingAbad, 2020-03-31:
Derdefout: Portmapping moest gedaan worden voor de Jetson Nano om over andere netwerken te kunnen verbinden

[image]

KingAbad, 2020-04-01:
Alle fouten en bijkomende errors opgelost en bij deze communicatie Jetson Nano Client met PLC Server werkt perfect 

zie video demo

[Demo_Com_PLC_Jetson.zip](https://github.com/KingAbad/Autonomous_Cart_2/files/4412055/Demo_Com_PLC_Jetson.zip)

KingAbad, 2020-04-01:
Hier is nog een demo waarop je kan zien dat ik de outputs kan aansturen via afgesproken commando's met het PLC team

commandos_VerlichtingToggelen.zip

closed 2020-04-03 by Fre101